spacetelescope/catkit2 #175: Discussion : make_property vs make_data_stream for scalar parameters

a-sevin, 2024-04-10:
I am pleased to see that catkit2 is now open-source.

I recently spent a few days installing it and developing my first service, proxy, and GUI for a hexapod. I will offer to integrate it when it is fully functional.

I have a question regarding when to choose a datastream over a property.

For instance, I began with the ThorlabsFW102C service using a datastream for its position. However, when dealing with a large number of parameters, using datastreams can become cumbersome as it may require a proxy.  

In contrast, the camera management service uses make_property, which is more elegant. 

Therefore, when managing a scalar parameter, it is more justifiable to use a property instead of a datastream.

RemiSoummer, 2024-04-10:
This reminds me of #51 and associated #89. 

The reason is that we wanted to have both the property and the data stream in some cases, e.g. to have the values accessible in the GUI (or other applications that may slip my mind).  

@ehpor will probably want to weigh in on this. 

And yes, we were excited to flip the switch!

raphaelpclt, 2024-04-10:
Hey!  (ah @RemiSoummer was faster than me to reference #89 )
I think the main reason is speed. Default properties are TCP communications between the processes (~ few ms), while datastreams enable few µs communication. That's also the reason #89 implemented datastream to manage most properties. 
It makes sense to have a slow property for the range of a motor, or the ROI of a camera. It's more relevant to have a datastream for the position of a motor that one might call at tens or hundreds of Hz for some experiments. 

Also yes, since the GUI is refreshing at ~10 Hz, it makes sense to have datastreams for the displayed values to avoid TCP communications for tens of values at this speed (even if the GUI can read properties as well).

raphaelpclt, 2024-04-10:
Also, the property you're going to create will very likely be implemented by a datastream in the backend. But it will benefit from the clean property interface.

closed 2024-04-11 by a-sevin