spacetelescope/catkit2 #189: Set up example use case and document

ivalaginja, 2024-04-24:


raphaelpclt, 2024-07-11:
I think that's a great idea! My intuition would be to keep all the example in one single contained location, like an ``testbed_example`` folder. I guess the goal is that someone new can copy this folder into a new repo, and start from that. This would be a "mini thd-control / hicat-package2" folder.

More specifically:
- Agreed that 1-3 go to a folder
- Wo do need a main window right now, but it shouldn't be too difficult to cut ties if needed. I wouldn't mind having a simple main window as an example too. 
- I would create a new very simple optical model too, to be in the example folder.

ivalaginja, 2026-02-17:
Superseded by #411.