spacetelescope/catkit2 #190: Fill in all service doc pages

ivalaginja, 2024-04-24:
- [ ] AimTTi PL-P
- [ ] AlliedVision camera - @ivalaginja 
- [ ] BMC DM
- [ ] Camera sim
- [ ] Empty service
- [ ] FLIR camera
- [ ] Newport Picomotor
- [ ] Newport XPS
- [ ] NI DAQ
- [ ] NKT Super K
- [ ] Omega ITHX
- [ ] Safety Monitor
- [ ] SNMP UPS
- [ ] Thorlabs CLD 101x
- [ ] Thorlabs filter wheel
- [ ] Thorlabs flip mount
- [ ] Thorlabs picomotor
- [ ] Thorlabs TSP temperature sensor
- [ ] Web Power Switch
- [ ] ZWO camera

steigersg, 2024-10-17:
Do we need to revise the bmc_dm documentation now that we have the new DM base class?

ivalaginja, 2024-10-17:
At some point probably, but I have added the note `(Old)` to that documentation and filled in the new doc page for the new services. So I think it's fine for now, but don't let me stop you if you want to update things.

closed 2025-03-12 by ivalaginja