spacetelescope/catkit2 #195: Check the first frame upon an exposure time change in Hamamatsu camera

ivalaginja, 2024-04-24:
> Question: does this faulty frame reappear when you change ROI, exposure time, gain or other settings?

So it turns out that we haven't paid much attention to the proper way of property setting with this camera - or the Allied Vision cameras. Some of the properties need the camera acquisition to be temporarily stopped, so I opened an issue for this for the AV service here: https://github.com/spacetelescope/catkit2/issues/183

For the Hamamatsu camera, only the exposure time can be changed while the camera is acquiring, the other properties need a temporary stop of the acquisition as well. Since that retriggers the acquisition loop, yes the faulty frame would appear again, but it would also be automatically discarded again in the current implementation. As for resetting the exposure time, I don't actually know how to catch the exact first frame after resetting it to figure out whether it's fine or not...

We only have our loaner camera for one more day and we won't be able to implement and test this before we have to send it back. We won't get our own camera for a while. My idea would be to open an issue an put a rework of the properties that need an acquisition halt into it (https://github.com/spacetelescope/catkit2/issues/194), akin to the issue for the Allied Vision cameras. And to open a separate issue to figure out how to test the first frame upon an exposure time change to decide whether it's fine or whether we need to throw it away too.

What do you say?

_Originally posted by @ivalaginja in https://github.com/spacetelescope/catkit2/issues/163#issuecomment-2074888363_

johanmazoyer, 2025-07-17:
@ivalaginja Do i still need to look at that ? We now ditch the first frame, I think it is enough (we don't restart camera often) and safe (we will never be perfectly safe that first frame is always correct)

ivalaginja, 2025-07-17:
I think there's some work left to do. One question from the original issue was whether we get a faulty frame each time when we start the acquisition, which would mean we get a faulty frame each time we change the ROI for example. This is because the frame width and height cannot be adjusted during an acquisition loop, so it stops momentarily for the parameter update and then restarts. So there might be a faulty frame each time this happens.

Since the exposure time doesn't need to interrupt the acquisition loop we don't think it's happening here, but it woudl be good to check too.

And finally, we do drop two frames each time we take images, but actually *only if the acquisition was already running*, for synchronization reasons. So if we have a faulty frame each time we start an acquisition, we're actually putting it on the image data stream and we should fix that.

johanmazoyer, 2025-07-17:
> I think there's some work left to do. One question from the original issue was whether we get a faulty frame each time when we start the acquisition, which would mean we get a faulty frame each time [...]

We ditch the first frame when we restart acquisition, so we are now ok with ROI change.

> Since the exposure time doesn't need to interrupt the acquisition loop we don't think it's happening here, but it woudl be good to check too.

We could also just ditch the first frame when changing the exposure time (without restarting) ? I'm afraid that it will be hard to be absolutely sure that it is working, as we only uncoutered the bad frames in a few accasion so it will be difficult to ensure that it will never happen. 
It will also ensure that we have the right time for this frame, since we are not sure when the frame was changed.

ivalaginja, 2025-07-17:
In this case, I think we're all good here, no further action needed. Feel free to close this issue if you're happy with things.