spacetelescope/catkit2 #210: Testbed to read service IDs from more than just `services.yml`

ivalaginja, 2024-05-31:
The testbed explicitly collects all service IDs found in `services.yml`:
https://github.com/spacetelescope/catkit2/blob/781ce3755b9db7a7abdae4945d58071bb7959ef3/catkit2/testbed/testbed.py#L176C41-L176C52

I am trying to come up with a way where I could have the testbed collect service IDs from more than just the file `services` configuration. In particular, we have so many individual motor services on THD2, that I would like to separate them out into a file like `motors.yml`. It seems like it should be easy to optionally pass additional config keys that would also be used as a source for service IDs. I am just still trying to figure out how to implement that without a change that is too invasive, and most of all keep things working with the current default setup.

ivalaginja, 2025-04-15:
@ehpor we're starting to transition * a lot* of motors into our testbed configuration, so I am coming back to this. Do you have any opinion on this to share? I am flying rather blind for now so any input is appreciated.

ehpor, 2025-04-15:
@ivalaginja Way back when you brought this up first, I was confused, and I still am now. Why is there a need for having multiple files instead of a single file? Is it just that the file becomes too long? All modern editors support code folding, which solves that.

ivalaginja, 2025-04-15:
Good enough for me. I agree with you and think this issue is obsolete.