spacetelescope/catkit2 #256: Compilation question on apple M3

axelpotier-astro, 2024-10-22:
I am trying to install catkit2 on my completely new apple M3 Mac. Have you guys ever seen the below error returned? It seems to be a local compiler problem. It tells me it cannot find the developer tools in the default location.

```
CMake Error at /opt/miniconda3/envs/catkit2/share/cmake-3.30/Modules/CMakeTestCCompiler.cmake:67 (message):
 The C compiler

  "/usr/bin/cc"

 is not able to compile a simple test program.

 It fails with the following output:

  Change Dir: '/Users/apoitier/Documents/Research/Softwares/catkit2/build/temp.macosx-11.0-arm64-cpython-39/CMakeFiles/CMakeScratch/TryCompile-2OThlF'
   
  Run Build Command(s): /opt/miniconda3/envs/catkit2/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_8c3f4/fast
  xcode-select: note: No developer tools were found, requesting install.
  If developer tools are located at a non-default location on disk, use `sudo xcode-select --switch path/to/Xcode.app` to specify the Xcode that you wish to use for command line developer tools, and cancel the installation dialog.
  See `man xcode-select` for more details.
   
   

  

 CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
 CMakeLists.txt:3 (project)
```

ivalaginja, 2024-10-22:
@RemiSoummer didn't you get a new machine recently? Did you successfully install catkit2 on it yet?

RemiSoummer, 2024-10-22:
You need to install Xcode and then open it once. Upon first time you open it installs the compilers and also there is a
License agreement to acknowledge.  

I also have a brand new M3 but haven't installed catkit2 yet. Won't be able to try that until Wednesday next week but i definitely will need to make it work.

alexisyslau, 2024-10-24:
I am not sure if this problem is related to M3 chip (?). My current machine is equipped with M3 Pro. First problem can be solved by installing the xcode, as @RemiSoummer suggested. 

I tried to install catkit2 once again by creating a new environment, successfully installed with no error. 

[image]
[image]

With this being said tho, some people encountered similar problem: 
[https://github.com/pybind/pybind11/issues/2229](url)

This is weird because catkit2 should already handle the installation of pybind11, as it is interfacing the C++ code with python ... Though can you find a directory called 'pybind11', located in /catkit2/extern? 

If not, I think you need to have get this folder by running ./download.sh at extern/

axelpotier-astro, 2024-11-06:
Thank you all,

Installing Xcode, removing and reinstalling the catkit environment worked out as a charm.

For information, I have just received the following warning when running 'python setup.py develop' :

`
CMake Warning (dev) at extern/pybind11/tools/FindPythonLibsNew.cmake:98 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  extern/pybind11/tools/pybind11Tools.cmake:50 (find_package)
  extern/pybind11/tools/pybind11Common.cmake:206 (include)
  extern/pybind11/CMakeLists.txt:200 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.
`

closed 2024-11-06 by axelpotier-astro