spacetelescope/catkit2 #266: Logging globally to a single file or folder

lanemeier7, 2024-12-02:
It would be nice to have a global log configuration or way to point all of the logging functionality to single consolidated file or folder (since multiple process architecture makes single file logs much more difficult.). This central location could have files broken up by day to keep files small and easy to find.

Thoughts? @ehpor @steigersg @ivalaginja @raphaelpclt

ehpor, 2024-12-02:
Carkit has a log distributor that collects log messages from all services and spits them out on a ZeroMQ PUB socket. The port number is 'testbed.logging_egress_port' or something like that. There is also functionality to listen to that port and write all log messages to a file. Look at the 'LogWriter' class. An instance of that class is to write the log for experiments.

However when outside of an experiment, log messages are ignored except for realtime display in the GUI when it is running. We didn't save them, because there was no clear indication on where to store them.

Also, normal stdout ('print') and stderr from services is lost. Also, because we didn't really find a nice way of storing it, but I put in the PIPE functionality in the 'TestbedServer' class to make it easy for someone to extend that and write it out to files.

So, as far as I can tell, a lot of functionality of what you're describing is already implemented. But maybe I'm misunderstanding?

lanemeier7, 2024-12-02:
Oh no, I am sure it is me who is misunderstanding. I was not seeing how the log files were saved. Can you point me to a minimal example?

lanemeier7, 2024-12-02:
It sounds like the hooks are all there but they don't do the storing part. Perhaps this issue could be about making the storage of log files an easy configuration change that is documented?

lanemeier7, 2024-12-03:
This was all a misunderstanding on my part. Now that I see how to make experiments by wrapping the Experiment class and that it makes the log messages error and all I can close this issue.

closed 2024-12-03 by lanemeier7