spacetelescope/catkit2 #267: Logging timestamps as default

lanemeier7, 2024-12-02:
Logging timestamps are useful references for experiments that run over long periods of time and trouble shooting between processes. It would be nice to configure catkit2's default to have timestamps for all log messages.

ehpor, 2024-12-02:
Timestamps are saved in the log files, but not printed in the testbed terminal. Honestly, the testbed terminal print was a debug print that I never bothered to remove. 😅

ehpor, 2024-12-02:
There are stored in the experiment folder, alongside the experiment metadata file and all output files from that experiment.

lanemeier7, 2024-12-03:
This was all a misunderstanding on my part. All log stamps are logged via the `LogWriter` class that is part of `Experiment`. In my test scripts where I was only using `ServiceProxy`'s and `TestBeds` I did not see this

closed 2024-12-03 by lanemeier7