spacetelescope/catkit2 #268: Propagate all service log messages

lanemeier7, 2024-12-02:
Currently, it is possible to services to crash and not show output when controlling services via a testbed and service proxy script. It would be nice to always propagate the output such that we can always see the info, warning and error messages and are not having to recreate the scenario by executing the service separately via the command line.

ehpor, 2024-12-03:
Is this a different issue from https://github.com/spacetelescope/catkit2/issues/266?

lanemeier7, 2024-12-03:
I think it still stands that services can crash and not propagate errors when used outside of an experiment. But it is much less important given that experiments can be made and in that context everything bubbles up and is logged. With that I am closing this. My apologies for the misunderstanding!

closed 2024-12-03 by lanemeier7

ehpor, 2024-12-03:
Well, the errors _are_ propagated, but not saved other than shown in a GUI if it's running.