spacetelescope/catkit2 #270: Thorlabs tsp01 service main expects 2 external temp sensors

lanemeier7, 2024-12-11:
The Thorlabs tsp01 temperature and humidity sensor can be configured with up to 2 external temperature sensors. When the temperature sensors are not used the service logs traceback error messages. In cases where none or 1 of the external sensors are used it will log a complaint every measurement check.

https://github.com/spacetelescope/catkit2/blob/825ce48c0ac79c7c7c6faae8f5fb250187c22987/catkit2/services/thorlabs_tsp01/thorlabs_tsp01.py#L25-L44

lanemeier7, 2024-12-11:
Perhaps I could make a configuration indicating which channels are being used.

Ex config entry for sensor1 and sensor2 which can be set to true or false if they are being used and want to be logged.
``` yaml
tsp01_1:
  service_type: thorlabs_tsp01
  simulated_service_type: thorlabs_tsp01_sim
  requires_safety: false
  serial_number: M00797061
  num_averaging: 1
  interval: 1
  sensor1: true
  sensor2: false
```

Any thoughts @ehpor @steigersg @raphaelpclt ?

ehpor, 2024-12-11:
Yes, definitely an oversight. You can make the required changes in a PR.

I'd go for:

```yml
tsp01_1:
  service_type: thorlabs_tsp01
  simulated_service_type: thorlabs_tsp01_sim
  requires_safety: false
  serial_number: M00797061
  num_averaging: 1
  interval: 1
  read_out_header_1: true
  read_out_header_2: false
```

I presume it also shouldn't make the data streams if the header is not read out.

lanemeier7, 2024-12-11:
yea I was also thinking it would not make the datastreams if we don't want the particular sensor read out.

PR #271 (2024-12-11): makes thorlabs_tsp01 temperature sensor heads optional per configuration
Closes #270 by making the Thorlabs tsp01 temperature sensor probes optional via services.yml configuration.

``` yaml
tsp01_1:
  service_type: thorlabs_tsp01
  simulated_service_type: thorlabs_tsp01_sim
  requires_safety: false
  serial_number: M00797061
  num_averaging: 1
  interval: 1
  read_out_header_1: true
  read_out_header_2: false
```
closed 2024-12-11 by lanemeier7