spacetelescope/catkit2 #281: Remove deprecated services `bmc_dm` and `bmc_dm_sim`

ivalaginja, 2025-02-02:
These have been superseded by the new classes/services introduced in #156. Both HiCAT and THD2 (still only in sim) use the new Boston services, and HiCAT has been using them successfully on hardware for over six months now. All functionality from the old services are included in the new services, except for #262, which is now being addressed in #280.

Would also need to delete their service doc.

@ehpor @RemiSoummer @raphaelpclt is there a good reason to keep these two old service classes or can I proceed with their deletion?

ehpor, 2025-02-04:
No real reason from my side. I'm gonna use a C++ service on the SEAL side.

ivalaginja, 2025-02-05:
@RemiSoummer @raphaelpclt opinions?

PR #285 (2025-02-07): Remove old BMC services
Fixes #281 after I got verbal approval from @RemiSoummer.

This removes the hardware service, simulated service, proxy, and their documentation. Also removes them from imports and the package entry points.

### Todo
- [x] Change the `bmc_dm` proxy used in testing: https://github.com/spacetelescope/catkit2/blob/9a8777eced0e5c5fe6809f84410e319011e43946/tests/config/services.yml#L4
closed 2025-02-07 by ivalaginja