spacetelescope/catkit2 #293: Installation on Ubuntu 24.04

rgalicher, 2025-03-11:
I followed the instruction in the README.md

Everything works up to :
conda env create --file environment.yml
conda activate catkit2

When I launch
pip install -e .

There is an error. See the screen shot below

[image]

rgalicher, 2025-03-11:
Here, is the complete output

[image]

[image]

[image]

[image]

[image]

[image]

raphaelpclt, 2025-03-11:
Ok we have a quick patch to solve this error - but a deeper solution will be required in the future.

Using an older gcc version solves it:

`sudo apt install gcc-11 g++-11` if you don't have it installed already
`C=/usr/bin/gcc-11 CXX=/usr/bin/g++-11 pip install -e .`

rgalicher, 2025-03-11:
Successfully installed catkit2-0.0.1
WARNING: There was an error checking the latest version of pip.

Thank you