spacetelescope/catkit2 #365: Trouble shutting down server

RemiSoummer, 2025-07-08:
When I did the testing for message broker on hardware I had an issue shutting down the server that requested more than one Ctrl-C.  I didn't think to much of it at the time because I was testing quickly and this can happen occasionally. 

We do not see this behavior on the simulator, so this might be due to windows or hardware or both. 

This is actually a systematic issue, and even just starting the server and closing it has a hard time closing some services and escalates through interruption to termination after a time out. This is with a single ctrl-C.  Eventually it manages to shut down, but this is clearly a new issue. 

First screenshot is with just the server running
Second screenshot is with server + GUI + science camera running. 

Note how safety and science_camera have trouble closing 

[image]

[image]

RemiSoummer, 2025-07-08:
FYI @ivalaginja in case you can test that on THD?

RemiSoummer, 2025-07-08:
adding that all the cameras seem to have the same issue so not a science_cam only problem.

ehpor, 2025-07-08:
This feels like a testbed-server issue with not updating the state of services correctly.

@RemiSoummer Are you able to check in task viewer that the process for the science camera was shut down after the `[science_camera] Service was closed.` message appears and before the testbed decides to terminate the process? Or is it still alive at that point in time and frozen?

RemiSoummer, 2025-07-09:
actually I got back to this and there were some left over processes that were not closed.  I ended up doing a couple reboots and now it's closing properly everything and I reproduced behavior a few times.  

Bottom line it was probably due to some left over process that didn't close properly at some point, now I don't know why this happened then but at least it's working okay when we are in a clean state.

closed 2025-07-09 by RemiSoummer